TonyM1958/HA-FoxESS-Modbus #1: More registers

canton7, 2023-04-20:
I'm going through your spreadsheet, comparing against what we have in foxess-modbus (thanks for that!)

StealthChesnut has the following holding register addresses which are missing from your speadsheet:

* battery_voltage: 11006
* battery_current: 11007

I might edit to add more tomorrow if I find them

TonyM1958, 2023-04-20:
Thanks. These are both present in the spread sheet, called InvBatVolt and InvBatCurrent with InvBatPower. These are the voltage, current and power sent to the BMS as measured by the inverter.

11034 / 11035 are Bat Volt and Bat Current. These are the voltage and current seen by the BMS when they arrive from the inverter.

Depending on whether the battery is charging or discharging, there is a small difference between InvBatVolt and Bat Volt. InvBatCurrent and Bat Current tend to the same, but the sign of Bat Current is reversed.

I tend to use the InvBatVolt, InvBatCurrent and InvBatPower in my calcuations and dashboard.

Let me know if you find any others - the unique_id is supposed to match Nathan's sensors and Stealthcheshnut as far as possible to simplify migration between integrations. Where unique_id is green, I think it's verified. Where its amber, I added a new unique_id as I didn't find an existing one that matched the sensor.

TonyM1958, 2023-04-20:
BTW, use the sort drop down under RS485. It will sort the sensors into numerical order 11000, 11001, 11002 etc

canton7, 2023-04-20:
Ah OK, let me look at those in some more detail later. Cheers

TonyM1958, 2023-04-20:
I've put the spread sheet on the wiki, so its easier to find: [H1-Modbus-Map.xls](https://github.com/TonyM1958/HA-FoxESS-Modbus/wiki/H1-Modbus-Map.xls)

TonyM1958, 2023-04-20:
To help complete the picture and stop this thread getting too long, I've renamed this issue to Battery Voltage / Battery Current. Please open a new issue for other sensors.

Looking at Stealthchesnut, modbusLAN: sensors 31020 / 30121, 31022 are BatVolt, BatCurrent and Battery-Discharge-Power. In modbusUSB, sensors 11006, 11007 are InvBatVolt and InvBatPower and 11008 is missing. In addition, 11034, 11035 are BatVolt and Temp respectively.

The problem is that 31020/31021/31022 and 11006/11007/11008 are the same sensors, presented by the inverter and are really InvBatVolt, InvBatCurrent and InvBatPower. The sensors 11034 and 11035 are only available via RS485 and are the BMS versions of Battery Voltage and Battery Current.

For consistency, this integration uses these names and ids:

11006 / 31020 = InvBatVolt, foxess_inv1_inv_battery_voltage
11007 / 31021 = InvBatCurrent, foxess_inv1_inv_battery_current
11008 / 31022 = InvBatPower, fox_inv1_battery_power
11034 = BatVolt, foxess_inv1_battery_voltage
11035 = BatCurrent, foxess_inv1_battery_current

InvBatPower is split by sign into the familiar 'Battery Discharge' and 'Battery Charge' by a template, retaining the same names.

canton7, 2023-04-20:
Right, I was too hasty and mixed up BatVolt and InvBatVolt I think. Thanks for the correction

closed 2023-04-20 by canton7