TonyM1958/HA-FoxESS-Modbus #2: Missing holding registers in range 32000 onwards

canton7, 2023-04-20:
Hi,

It looks like the input registers 11069 to 11092 have corresponding holding registers in the range 32000 to 32023

TonyM1958, 2023-04-21:
Thanks - I've added these to the spread sheet. Will do sa scan later to verify and also see if these are the same on KH series if possible.

canton7, 2023-04-21:
They work OK on my H1 when accessed over RS485 reading the holding registers. I don't have a lan port to test. 

OK, interesting...

canton7, 2023-04-21:
That will be the first instance where the set of holding registers available over LAN is different to that available over RS485... Are you sure they don't work over LAN?

TonyM1958, 2023-04-21:
Need to have a poke around and see what's going on. I think I can see the registers when I poll them directly via the LAN port, but it seems HA doesn't see them - not sure why right now. Have not, yet, tried to access holding registers over RS485 instead of input registers. All the input registers seem to work ok.

TonyM1958, 2023-04-21:
Ahh - might help if I changed the input_type to holding when copying the sensors from RS-485 to LAN as well... need to go through and check all the sensors are setup correctly....

canton7, 2023-04-21:
Ah phew! Yeah the 3xxxxx registers are always holding, whether they're accessed over lan or RS485.

TonyM1958, 2023-04-23:
updated my code to read these registers and also aligned the different connection methods so only one modbus file is now required for LAN and RS485 connections for single phase inverter H1, AC1, AIO and KH series

closed 2023-04-23 by TonyM1958