TonyM1958/HA-FoxESS-Modbus #4: Input registers for inverter temp / ambient temp are the wrong way around

canton7, 2023-04-20:
On my H1, I get 11024 and 31018 giving the same value, and 11025 and 31019 giving the same value.

Your spreadsheet has them the other way around.

TonyM1958, 2023-04-21:
Side by side compare of my data with FoxESSCloud:

[image]

As you say, my data here is coming from RS485, so my values are reversed - many thanks for pointing this out, I'll correct in the spread sheet and my code.

(think I just assumed it didn't make sense for the inverter temperature to be lower than the ambient temperature so swapped them! There has always been a bit of a mystery as to why these are labelled as they are. You could argue that Fox has got them the wrong way around, but it's best to match the Fox data)

TonyM1958, 2023-04-21:
Updated the code, but it wasn't the change I expected. This is the way round that makes my sensors match FoxESSCloud:

31018 / 11024 = InvTemp
31019 / 11025 = AmbTemp

Does that now agree with your understanding / data?

canton7, 2023-04-21:
Yep, I agree with that :+1:

I was just going by the work that you and others have done on fixing which of 11024 and 11025 is InvTemp/AmbTemp. When I was sorting out the holding registers, I noticed that 31018 had the same value as 11024, yet your spreadsheet said it was AmbTemp.

(I'm using holding/input vs LAN/RS485 to refer to the different register types, mainly because the H3 doesn't expose any input registers as far as we know, so you need to access the holding registers over RS485 as well...)

TonyM1958, 2023-04-21:
For info, KH is a mid point between H1 and H3: it is a single phase inverter, without an ethernet port but with RS485. When you read the RS485 via W610, you find there are holding registers (3xxxx) rather than input registers(1xxxx). So, when you use this with the modbusLAN profile, you see all the holding registers you would expect to see via the ethernet port on H1 i.e. single phase register set.

canton7, 2023-04-21:
Ah fun, like the H3 in that respect. (note that everything exposes holding registers. It's just that some things also expose input registers).

What's the serial at register 30000 onwards, do you know? I need that to auto-detect the model

TonyM1958, 2023-04-21:
75, 72, 49, 48, 46, 53 = 'KH10.5'

canton7, 2023-04-21:
Thank you so much! How annoying they didn't follow the normal pattern of putting a dash after the model...

closed 2023-04-21 by canton7