TonyM1958/HA-FoxESS-Modbus #5: More version/model registers

canton7, 2023-04-21:
I'm not sure if you care, but in the name of completeness...

1. Inverter type (e.g. H1-3.7-e) is at 10000/30000, for 10 (?) registers
2. 10016/30016: master version
3. 10017/30016: slave version
4. 10018/30016: manager version

TonyM1958, 2023-04-21:
Yes, I can read these and get values back from 30000 - 30023 from the H1 ethernet port and RS-485 port:

[image]

The inverter model looks to be the first 16 registers, encoded as 16-bit ASCII, so this reads 'H1-6.0-E        '. Haven't worked out an easy way to read this yet. Thought setting data_type: string and count: 16 might work, but this returns unknown.

The versions are present and correct.

No idea what the values are for registers 19-23.

canton7, 2023-04-21:
Yeah, we use the inverter type registers in foxess_modbus to auto-detect what inverter the user has on connection. I think the H3 yaml in StealthChesnut's repo has an example of reading them with the modbus intigration?

19 looks like it might be slave id? (somewhat irrelevant since you need to know the slave id to read that register...)

TonyM1958, 2023-04-21:
I think I need a setting for char encoding. It complains in the log that the text isn't valid utf-8 which is correct as ut either utf-16 or unicode. However, it won't accept a setting for encoding, so I'll have a look in the other repo.

19: similar, but wouldn't this be 247 if it was the slave address, not 257?

canton7, 2023-04-21:
Good point! It's not slave id then...

It is utf-8, as utf-8 is a superset of ASCII, and it's ASCII. It's not utf-16 that's for sure (every 2nd byte would be 0 in that case). 

What's your sensor config and error exactly?

TonyM1958, 2023-04-21:
I'm currently looking at Input Register 100000. If I set count:14 or 15, I don't get the error. it returns the inverter model and then the inverter serial number... like this: 'H1-6.0-E65BH602026EB00'. That is the serial number but mine ends with 4 added, so I'm begingin to think this is actually 2 fields with 8 characters for the model and 12 characters for the serial number?

canton7, 2023-04-21:
I wonder whether that might vary by inverter type. The AC1 starts with "AC1" and APO with "AIO-H1"

TonyM1958, 2023-04-21:
I can read the model, as long as I always set count to 1 less than the length. For example: count:7 works, but count:8 gives this error:

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/modbus/sensor.py", line 119, in async_update
    result = self.unpack_structure_result(raw_result.registers)
  File "/usr/src/homeassistant/homeassistant/components/modbus/base_platform.py", line 197, in unpack_structure_result
    return byte_string.decode()
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x81 in position 15: invalid start byte
```

But then, I always lose the last character, so the model reads 'H1-6.0-'

The serial number is almost as wierd. I read count: 7 and get the value: '65BH602026EB00'. Again, last char missing but the string length is actually 14 characters, like this was byte encoded.

canton7, 2023-04-21:
In case it's useful: https://github.com/StealthChesnut/HA-FoxESS-Modbus/blob/main/custom_components/HA-FoxESS-Modbus/modbus-H3-LAN.yaml#L452

TonyM1958, 2023-04-21:
> 
> I've just noticed that the final registers are 0x0101, 0x1010, 0x2010, 0x3010, 0x4010 in hex...

So, that looks like the battery firmware versions using a digit each for M.1.0.1, S1.0.1.0, S2.0.1.0, S3.0.1.0 and S4.0.1.0 as displayed on the front panel for the BMS and 4 x HV2600 that I have. As you can install up to 8 batteries, I would expect the next 4 registers to be reserved for the next 4 batteries.

TonyM1958, 2023-04-21:
You might need to check this out, see if you get the same as me.... all over RS485.

Reading input 10000 as string, the max I can get without error is 7 chars. Reading holding 30000 as string, I can read 16 characters without error and get the full model number right padded with spaces.

Reading holding 30008 as string returns blank. Reading input 10008 as string, I can read 7 characters without error and I get back the first 14 chars of the inverter serial number. Not found a way to read the last character without error yet.

So, that's why the H3 code works but you don't get the serial no. I suppose I can setup to read the model via holding 30000 and most of the serial no over input 10008...

canton7, 2023-04-21:
I'll give that a shot tomorrow.

I'll check the battery versions. My batteries haven't arrived yet, so in theory should all be 0 for me...

canton7, 2023-04-22:
Yeah, 30019-30023 are all 0 for me, which makes sense given my lack of battery.

10000-10007 look like ASCII for me -- "H1-3.7-E". 10008 is the much larger value 13872.

30000-30007 are the same "H1-3.7-E", but 30008 to 30012 are ASCII 32, or spaces. 30013 onwards are all 0.

[10000]: 72
[10001]: 49
[10002]: 45
[10003]: 51
[10004]: 46
[10005]: 55
[10006]: 45
[10007]: 69
[10008]: 13872
[10009]: 16968
[10010]: 13111
[10011]: 12848
[10012]: 12857
[10013]: 17986
[10014]: 12339
[10015]: 13824
[10016]: 148
[10017]: 102
[10018]: 145
[10019]: 0
[10020]: 0
[10021]: 0
[10022]: 0
[10023]: 0
[10024]: 0
[10025]: 0
[10026]: 0
[10027]: 0
(Then invalid frames)

[30000]: 72
[30001]: 49
[30002]: 45
[30003]: 51
[30004]: 46
[30005]: 55
[30006]: 45
[30007]: 69
[30008]: 32
[30009]: 32
[30010]: 32
[30011]: 32
[30012]: 32
[30013]: 32
[30014]: 32
[30015]: 32
[30016]: 148
[30017]: 102
[30018]: 145
[30019]: 0
[30020]: 0
[30021]: 0
[30022]: 0
[30023]: 0
[30024]: 0
[30025]: 0
[30026]: 0
[30027]: 0
(Then illegal addresses)

TonyM1958, 2023-04-22:
Try reading 10008 as string, count 7. On mine, this is ascii bytes that return your inverter serial number.

TonyM1958, 2023-04-22:
According Bob Dopper on the FB group, he used to work for Solax and Fox ESS , licensed the H1 inverter design from Solax. This is why people comment that the front panels look so similar. The LAN modbus port is a legacy of this, which is why Fox says its not supported. Fox is only developing new features around RS485 andthis is why some new registers are now available on RS485 but not on LAN and possibly why LAN only has holding registers?

I'm sure there are grains of truth in there somewhere...

canton7, 2023-04-22:
Ah, so 2 ASCII chars per register. I'll try doing that later.

Oh interesting... I was trying to find a modbus spec for foxess hiding in a corner of the internet, but maybe I should have been looking for solax!

Also wonder how that ties in with the H3, which apparently only had holding registers.

canton7, 2023-04-22:
Yep, 10008 to 10015 are my serial number. Note that the least-sigificant byte of 10015 is 0.

canton7, 2023-04-22:
Some of the register names here look familiar, but the addresses don't line up in any obvious way

TonyM1958, 2023-04-23:
updated my code to load these values and firmware versions.

closed 2023-04-23 by TonyM1958

TonyM1958, 2023-04-23:
BTW, just wanted to check. My serial number is 14 chars from 10008, but its 15 chars in the app with 4 appended to the end.

Is yours the same?

Just wondered if the last char in the app might be a country code, 4 = UK?

canton7, 2023-04-23:
My serial is 15 chars from 10008 - only half of 10015 is used. I can't currently log into the app to check due to yet another foxess outage...

TonyM1958, 2023-04-23:
may be it's a data error when the serial number was loaded on my inverter... if I use count: 8 reading the serial number, I definitely get this error:

```
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x81 in position 15: invalid start byte
```

If I use count:7, it works but returns this: 65BH60xxxxEB00. My serial number ends EB004.

canton7, 2023-04-23:
What's the value of your registers 10013-10015?

TonyM1958, 2023-04-23:
the 15th char is there, just looks like modbus can't manage to read it because the 16th byte is illegal. The raw bytes are:

69, 66, 48, 48, 52, 129 = EB004?

So, the last byte is screwing up the string decode, 0x81 just like the error message says. Would be rubbish if you had to resort to reading 8 x uint16 and building the string manually... but I don't see how you can read an odd number of bytes from 16 bit registers

TonyM1958, 2023-04-23:
wonder isn't corrupt but something to do with battery status, you can load the serial number as it's 0x00 for you with no batteries. I have 0x81 and I have 1 x BMS and 4 x batteries...

canton7, 2023-04-23:
It's weird that they'd stick that into a serial number register...

canton7, 2023-04-23:
I wonder how this differs on other inverter models. 8 chars is enough for a H1, but "AIO-H1" is a bit longer...

TonyM1958, 2023-04-23:
Yes, also wondering what will happen when they change the model names. Did you see they have submitted type tests for H1-6.0-E-G2...

canton7, 2023-04-23:
Yeah I saw that.... Just need to find someone who has one now!

TonyM1958, 2023-05-22:
> wonder if byte 16 isn't corrupt but something to do with battery status, you can load the serial number as it's 0x00 for you with no batteries. I have 0x81 and I have 1 x BMS and 4 x batteries...

Reading serial number from 10008 on my H1 is now working when last month it has a spurious character in byte 16. I've added 2 more batteries, taking the total to 6, but haven't changed anything else. Anyway, I can reliably read string(8) from both 10000 and 10008 and get both model and serial number now...?

canton7, 2023-05-22:
Huh, interesting! I wonder whether there's another bit of information encoded in there... I'm going to keep reading from 30000 anyway, as it seems a bit more reliable.