TonyM1958/HA-FoxESS-Modbus #6: Search for inverter fault codes

TonyM1958, 2023-04-23:
H3 profile has addresses for the inverter status and 3 x fault codes. These must also be present for H1, but aren't documented so far as I know.

TonyM1958, 2023-04-24:
H3 status code / fault code is 30041, 30044, 30045, 30046.

H1 has registers in range 30030 - 30039 - expect status and fault codes are in this range. First guess is 30031, 30034, 30035, 30036

canton7, 2023-05-04:
Are you sure about 31031 etc? My inverter reports a fault because it doesn't have batteries connected yet, but those registers are all 0

TonyM1958, 2023-05-04:
Definitely read up to 31039 via LAN as holding registers on my H1 without getting illegal address. I'm currently running from RS485, also reading 31031, 31034, 31035, 31036 as holding registers.

HA is showing these are all zero but not unavailable (as they would be if the inverter was saying these are illegal addresses):

[image]

Obviously, no idea what these registers might actually contain... Still not something I would put into Nathan's code until they are properly identified? HTH

canton7, 2023-05-04:
They don't have to contain anything. For example, the implementation might just be an array of uint16s, one for each register. You'd just cover a range of the address space with your array (so eg address 30000 maps to array index 0). If there are big gaps in the addresses you want to implement you might have two or more arrays, but you wouldn't do that for a few uninplemented registers: you'd just map them to an array element as normal. Any array elements which you don't use will just contain 0.

canton7, 2023-05-04:
... And if they did raise an invalid address exception there, it would stop you people doing bulk reads.

TonyM1958, 2023-05-05:
Inverter State Code = 31027 on H1 updated in spread sheet. Decode to Inverter State / foxess_inv1_state via template:
0 = Stopped, 1 = Checking, 2 = On Grid confirmed. Other values to be confirmed, possibly 3 = EPS (Grid Lost), 4+ = Fault (based on info from H3).

No fault code addresses found yet, but values of 1 are present in 31028, 31029 on my H1 inverter. Will need to generate a fault to see what happens...

canton7, 2023-05-05:
Nice work!

TonyM1958, 2023-05-07:
31025 is the holding address for input 11041 BMS Max Charge Current and 31026 is the holding address for input 11042 BMS Max Discharge Current (which may mean these values are available for both LAN and RS485 connections). 

These appear to be the maximum charge / discharge current that can be delivered to the BMS / batteries based on their current state and are different to the Max Charge and Max Discharge Current settings (41007/41008). For example, when the batteries are full, BMS Max Charge Current reduces to zero to stop the charge current.

Not tested, but might expect BMS Max Discharge Current to reduce to zero if the batteries are at Min SoC?

TonyM1958, 2023-05-10:
Input register 11013 identified as the RPower In. This tracks but is always slightly higher than RPower and is also unsigned. For example, when force charging the RPower In remains +ve, whilst RPower is -ve.

Input register 11019 identified as EPS RPower In. Same as above but related to EPS.

These don't appear to add significant value over the RPower and EPS RPower sensors already in use.

TonyM1958, 2023-05-12:
Input register addresses now available in spread sheet, based on PDF. Code updates still to do to reflect the changes and map holding registers where possible.

closed 2023-05-28 by TonyM1958