TonyM1958/HA-FoxESS-Modbus #7: Add H3 register values

TonyM1958, 2023-04-23:
Plenty of reference data around for H2 holding registers. Should modbusH3 file be added?

TonyM1958, 2023-04-24:
Added modbusH3.yaml and spread sheet tab - needs to be tested by someone with H3 inverter. 
Modified generic template.yaml so it works with both single phase and three phase inverters.

cosmico89, 2023-08-08:
HI!
I own a H3.10kw, if you need to test something I can try! 😄

TonyM1958, 2023-08-08:
Thanks @cosmico89. What's your current setup in terms of home assistant and monitoring? Don't want to disrupt stuff too much, so any context will help. There are definitely some registers and behaviours I'm not certain about.

cosmico89, 2023-08-08:
i am currently using this
[https://github.com/StealthChesnut/HA-FoxESS-Modbus](https://github.com/StealthChesnut/HA-FoxESS-Modbus)
being that it is no longer supported I wanted to evaluate which one offered the most information
mine looks very well done.

i was going to try
[https://github.com/nathanmarlor/foxess_modbus](https://github.com/nathanmarlor/foxess_modbus)
but maybe it seems to me that yours has more information or am I wrong?

thanks!

TonyM1958, 2023-08-09:
Thanks - it's as much about how comfortable you are tinkering with code and seeing if and how things work as it's about equipment. Here's some background.

The StealthChesnut integration was ground breaking for lots of people in terms of getting in HA with their solar, including me. However, quite a lot of the setup was in configuration.yaml and some people struggled with this, especially if they needed to update things.

I took this integration and changed the way configuration.yaml was structured to make it easier to setup and maintain and then added and evolved this into the integration I have today - this github site is basically a backup of my running config.

Nathan took a different approach and re-wrote the integration so it became a fully fledged and independent HA integration. This allows the configuration to be automated so it's easier for most people to get installed.

I guess the differences are, my integration is a home grown kit car that requires you to be comfortable editing files to setup and maintain but that is open and can be changed, fine tuned and added to. In my view, it a dev and test platform that allows me to experiment and document more features as we find them or need them. I don't use version control but I try to maintain a reasonably good stucture and track changes through github. Someone with KH contacted me to get my integration working, so we did some digging around and added this. I copied info about H3 for other integrations but I still don't think we have a complete set of registers or behaviours. What we know is documented [here](https://github.com/TonyM1958/HA-FoxESS-Modbus/wiki/Modbus-Map.xls) and my integration is a running proof of concept for this.

Nathan's integration is a production car - lots of people install it and run it, it has version control so you can update or roll back. However, it's harder to change what it does or to add to it - you need to put in feature requests or bug reports so they can develop updates. Originally, Nathan only really supported H1 but they have now added support for KH and H3 building off the work we did documenting stuff through playing with my integration.

I've added stuff that isn't in Nathan's integration (yet?). I think there may be one or two people with H3 running my code but it's public so if someone downloads it and doesn't contact me I don't know.

My suggestion would be first, do a full backup. **If** you are up for experimenting, try installing my code and see how you get on - I'll help where and when I can. If it doesn't work anything like you need, tear it down and install Nathan's code. If you don't want to experiment, install Nathan's code anyway.

There are some issues maintaining history whenever you change integrations but but these can usually be solved.

cosmico89, 2023-08-09:
Thanks so much for the super detailed explanation!!!
then if I want I can also create a separate server with a dedicated RS485
the simple question is... can i connect 2 RS485 and read the data together? so that I can use the current version of HA in production and test safely and help you make your code work better

TonyM1958, 2023-08-09:
RS485 has 1 master with multiple slave devices. HA operates as the master to get data from your inverter so you can only 1 HA device connected. HA could poll more than 1 inverter via RS485 but not the other way around. Sorry. You do still have the Fox data logger and cloud but, obviously, that's not the same.

cosmico89, 2023-08-09:
I am currently connected with RS485, I have the Wifi DataLogger associated with the cloud

TonyM1958, 2023-08-09:
Bit more info... I've basically taken the known register set for H3 and added in new registers we found on H1 and KH but these have not been tested. So one thing would be to load up the sensors and check for illegal address errors in the HA log where modbus cannot read what we have. As part of this, review all the sensor values and confirm they read OK, including decodes and model id for inverter, bms and batteries.

Another area is we have 8 x fault registers on H1 but only 3 on H1. With no fault, they read 0 so are hard to detect unless the addresses flag as illegal. So, add some more and see if they read OK.

Final area, after whatever gets flagged above is how Load Power is calculated. This needs to be validated against Fox cloud data.

 Ideally, we would also check load power results with a second solar inverter connected via meter 2 but that's not easy or common so I'm not expecting to be able to do that.

cosmico89, 2023-08-09:
ok installed and configure with all correct parameters for H3
I note that the data of the following entities are not available (on HA):
I upload a long photo let me know if it's convenient for you, otherwise I'll start writing :smile: 
[image]

TonyM1958, 2023-08-09:
Thanks - that's quite a list so will take a while to check over. Can you highlight (name) any sensors you have in your existing dashboard that have disappeared? I'll look at those first. Also, you view on how functional what you have now is compared to what you are used?

How many and what sort of batteries do you have? What firmware version (haven't checked, it might tell me above?)

TonyM1958, 2023-08-09:
Please can you also post a copy of home-assistant.log that shows what happened after you rebooted?

TonyM1958, 2023-08-09:
There is a lot of duplication in the list of sensors, like everything might be installed twice. Please can you have a look and check if the duplicates are just listed twice in that dashboard view or if they are using different sensor names?

Please can you also look at Settings, Devices, Entities and search using _2. The only sensor that matches should end fault_2. Any others need to be cleaned up by removing old names that are no longer used and renaming the new sensors without _2 at the end. Send me any examples if you are unsure.

Might be easier to look at stuff in Messenger if possible as chatting through comments is a bit slow, assuming we can work through any translation issues?

cosmico89, 2023-08-09:
I proceed with everything you asked me ... do you happen to have Telegram?

TonyM1958, 2023-08-09:
I assumed you were in the Fox ESS FB group and could drop me a Message. Is that OK?

TonyM1958, 2023-08-09:
Yes, try this post: https://www.facebook.com/groups/foxessownersgroup/permalink/1908006026248237/

Click on the photo and it show a screen that has a message button. If you are in the FB group as well, you should not be blocked.

TonyM1958, 2023-08-10:
Scanning 31000 range, registers are as expected up to 31040. Then:
+ 31041 Inverter State Code
+ 31042 unknown = 1
+ 31043 unknown = 1
+ 31044 Fault 1 code
+ 31045 Fault 2 code
+ 31046 Fault 3 code
+ 31047 Fault 4 code
+ 31048 Fault 5 code
+ 31049 Fault 6 code
+ 31050 Fault 7 code
+ 31051 Fault 8 code

That's the end of that block of registers

TonyM1958, 2023-08-10:
Scanning 41000 range, the holding registers are:
+ 41000 Inverter Work Mode
+ 41001 - 40116 not available, Charge Time Periods (read as a block?)
+ 41007 Max Charge Current
+ 41008 Max Discharge Current
+ 41009 Min SoC
+ 41010 Max SoC
+ 41011 Min SoC On Grid
+ 41012 not available
+ 41013 not available
+ 41014 unknown 1
+ 41015 not availabe
+ 41016 unknown 4621
+ 41017 unknown 0
+ 41018 unknown 0
+ 41019 unknown 4621
+ 41020 unknown 0
+ 41021 unknown 0
+ 41022 unknown 3171
+ 41023 unknown 1847
+ 41024 unknown 16

TonyM1958, 2023-08-10:
[image]

This is what we see scanning holding registers from 44000. Looks like:

+ 44007 Power Out Limit
+ 44008 Power In Limit

Others TBC - this block may be related to input registers 11026 - 11033 on H1?

TonyM1958, 2023-08-10:
[image]

Expected Inverter Model, and Firmware versions at holding register block 30000. Model number has leading and trailing spaces.

TonyM1958, 2023-08-10:
Power calculations: meters are combined from raw data and not split by phase, Template calculations updated to reflect this and now appear to agree with Fox Cloud data.