TonyM1958/HA-FoxESS-Modbus #8: Search for more inverter controls

TonyM1958, 2023-04-28:
All the values that can be set on the inverter via the installer mode app must be able to be set via RS-485. I know I can load the min and max soc and work mode values etc. There must be more values you can read and potentially write such as max charge current etc.

Needs a search through the input registers in 4xxxx range?

canton7, 2023-04-29:
https://github.com/StealthChesnut/HA-FoxESS-Modbus/wiki/RS485-Agent-Addresses

TonyM1958, 2023-04-30:
thanks @canton7, good list. I've picked up a few of these, verified them against my H1 and am adding them into the spread sheet as a central reference along with data_type, scale, units and precision. Also added into my code base for testing.

TonyM1958, 2023-05-14:
updated the modbus and template files for main inverter settings so that codes read from the inveter can be loaded and displayed. Some codes only partially decoded, such as sesnor.grid_code to sensor.grid_safety:

+ 2 = G98_UK
+ 3 = G99_UK

Other values will need to get provided by people with inverters in different regions / settings.

closed 2023-05-14 by TonyM1958