TonyM1958/HA-FoxESS-Modbus #16: Update always reverts connection settings in config to USB

m24andrew, 2024-02-28:
When I update the integration via hacs the modbusH1_RS485.yaml always reverts back to USB connection despite it being set to LAN connection before the update. When I restart HA I get an error when checking the configuration and I have to comment and uncomment the relevant settings every time.

[image]

TonyM1958, 2024-02-28:
Thanks - that's expected behaviour as it reverts the file to the initial state before you edited it, but I agree that it is not desirable.

I've updated the integration to resolve this by duplicating the modbus files with the suffix _LAN so they are configured to use the LAN adapter instead of USB.

You can simply rename modbusH1_RS485.yaml to modbusH1_RS485_LAN.yaml and update your configuration.yaml to refer to this file, if you wish.

Once you make this change, any update via HACS will keep your chosen connection method.

m24andrew, 2024-02-28:
Amazing, that has worked a treat. Thanks for such a rapid update. 😃

closed 2024-02-28 by m24andrew