TonyM1958/HA-FoxESS-Modbus #21: FoxEss Modbus Integration: FoxEss H1 Inverter Settings

IanR007, 2024-03-22:
In Inverter Settings "communications" ensure Ethernet is set to "DHCP". This reveals 3 settings which must match your Modbus logging device set up as follows;
IP (LAN) Address: 10.10.100.254
Subnet (Mask) Address: 255.255.255.0
Gateway (DNS): 223.5.5.5

TonyM1958, 2024-03-22:
Not sure if you posted this comment in the right place? Note that the Ethernet port is disabled by firmware on H1 with Master 1.70 so it will not work however you set it up. You need to use RS485.

IanR007, 2024-03-22:
Tony,

 

I’ve just sent you an email to explain the problem I encountered.

 

Trust this helps?

 

Kind regards,

 

IAN RAWLINSON

1 Slaley

Fatfield

Washington

Tyne and Wear

NE38 8PG

Tel: 0191 4169642

Mobile: 07763 359817

IanR007, 2024-03-22:
Tony,

 

PS: It could be something to do with firmware updates Fox have applied to the Inverters?

 

Regards,

 

Ian

TonyM1958, 2024-03-22:
I've edited your comments as this repository is public and it exposed your contact details.

I think there is a problem with your setup based on what you are saying. It sounds like you have a network cable plugged into the inverter Ethernet port instead of into the ELFIN adapter. That is the only reason I can think of why you need to set the Ethernet port address on the inverter to get it to work?

This port will not provide all the data you will get via RS485 using the ELFIN adapter.

Please can you check your setup?

IanR007, 2024-03-22:
Tony,For your info I'm getting ALL of the data from my EW11.The problem I had was no communication which prevented your automatic installation.The Inverter was set to RS485 247, however it would still not communicate until I went into Ethernet and set it to "disable" it then revealed the info I've put in my email.Once I'd done that the device (EW11) started communicating with the Inverter.I can only relay my experience as this is very much a learning curve for me, until I got it working every obstacle became another hurdle to overcome.By the way I'm using a stand alone HA server.RegardsIanSent from my Galaxy

IanR007, 2024-03-22:
Tony,I can send you photos plus screenshots if you want them as proof?Cheers,IanSent from my Galaxy

IanR007, 2024-03-22:
Tony,Would appear I have found a hack to kick start an Elfin Ew11 to talk to the Inverter. I've been back in the settings and set "disable" for the ethernet and the device is still communicating with the inverter.Apologies if I have caused any inconvenience? This is ALL learning for me!Have a good night.Best regardsIanSent from my Galaxy

TonyM1958, 2024-03-22:
I believe you, I just don't see  how this could stop the installation of this integration as it just copies files to HA. The only configuration is setting the IP address of the adapter in 'secrets.yaml'.

I understand the ELFIN network is quite tricky to configure. I piggy-backed the info from Nathan's integration here: https://github.com/nathanmarlor/foxess_modbus/wiki/Elfin-EW11.

I can't easily update these instructions as they belong to someone else. Were these instructions you used?

IanR007, 2024-03-22:
Tony,I followed your latest update which incorporated the latest fixes. I've been trying for 5 days with the original integration were you upload everything through file editor. Once I'd managed to load HACS your automation was like a breath of fresh air because I felt I was so much closer to getting it to work. It still blocked me 3 times until I did the settings on the inverter. Once I'd done that it loaded no problems.PS: I had to do some gardening in between as it did start to frustrate me.Cheers,IanSent from my Galaxy

FozzieUK, 2024-03-22:
I’ve seen it before Tony, it’s not your integration - it’s the inverter, but only when it has been communicating using the old LAN port does it stop the RS485 from working - it’s not obvious as the Inverter comms and slave all appear to be set correctly, but going in and disabling the LAN port restarts the Manager and then it works ok on RS485. I guess less of a problem going forwards now the LAN functionality has been disabled.

TonyM1958, 2024-03-22:
Thanks. Disabling the LAN port makes a bit more sense. However, I think any changes to the network settings on the inverter reset the network stack so might also re-enable RS485. I used to run both RS485 and LAN and switch between them and haven't seen this.

In summary, it probably doesn't matter what settings you use as long as you make a change that causes the network stack to reset.

TonyM1958, 2024-03-22:
Actually, I have seen this before. Using the inverter LAN port crashes the Manager. The normal symptom is that modbus responses stop. I had not realised this also stops RS485 modbus. Obviously, changing the network settings, restarts the manager process. The side effect of this is that it also restarts RS485.

All the reasons why Fox disabled the LAN poet in a nutshell!