TonyM1958/HA-FoxESS-Modbus #24: connect issue

halfprice11, 2024-07-23:
Hi,

tried to connect with your integration to my Fox inverter.
But not yet with success.

Log file tells me, I do have connection problems:
2024-07-23 11:31:10.458 WARNING (MainThread) [pymodbus.logging] Failed to connect [Errno 111] Connect call failed ('192.168.1.157', 502)

I am wondering, if I do need to set any password and user, to connect to my USR-W610 via LAN?

Any help appreciated
Regard

BTW: I do have the FoxESS - Modubus integration up and running in parallel, but do not see any overlapp and problems, doing so.
This Integration does work, but does not handle the Fox inverter update 2.0 well, mixing up feed in and grid consumption

halfprice11, 2024-07-23:
found the mistake. simply did use TCP instead of UDP.
Sorry for the spam.

closed 2024-07-23 by halfprice11

TonyM1958, 2024-07-23:
No problem. Please can you describe where you made the change, just in case this helps anyone else with the same issue. Thanks

halfprice11, 2024-07-25:
sure.
did double check the settings within my modbus - LAN (USR-W610) if it is setup for UDP or TCP mode.
Second, I did open your custom_componats/HA-FoxESS-Modbus/modbusH3_RS485_LAN.yaml
line 25 should have some type as the USR ..... UDP or TCP (it comes with TCP after first installation)

:-)

TonyM1958, 2024-07-25:
Ok, just be aware, if you update via HACS, it will over write the file under custom_components and your changes will be lost. It will also remove any files you create in custom_components so, if you create a copy of the file, best to put it under CONFIG.